maz0r/klipper_canbus #9: How can i connect Mks monster 8 v2 by Can Bus?

quocanhcgd, 2022-11-08:
How can i connect Mks monster 8 v2 by Can Bus?
Direct to usb on PI or by UTOC?
If direct, i will connect  CAN_L and CAN_H to USB on PI or TX and RX to GPIO on PI?
Thanks

maz0r, 2022-11-08:
The monster8 can be connected over CAN in only three ways I am aware of.

USB PA11/12 in can bridge mode (acting as the network bridge and a can node concurrently).

USB PA11/12 as a CAN device (with a UTOC acting as the network tranciever) using RX/TX signalling.

CAN PA8/9 as a CAN device acting as its own tranciever reciwving H/L signalling .

closed 2022-11-08 by maz0r

quocanhcgd, 2022-11-09:
303 / 5,000
Translation results
Can you have a detailed picture of the wiring in each situation?
If I use PA11/PA12 USB bridging mode, how do I configure it?
I would love to not use UTOC but connect directly to the PI. Which I don't really understand how to connect.
Mind you because I'm a bit poor at this.
Please help me.