maz0r/klipper_canbus #22: ERROR:root:Can Read Error

Johnnyyr, 2022-12-09:
I think that is not an issue, but can you tell me something about this error? thx

`[##################################################]

Write complete: 13 pages
Verifying (block count = 413)...

[#########ERROR:root:Can Read Error
Traceback (most recent call last):
  File "/home/pi/CanBoot/scripts/flash_can.py", line 137, in send_command
    ret = await self.node.readuntil()
  File "/home/pi/CanBoot/scripts/flash_can.py", line 287, in readuntil
    return await asyncio.wait_for(self._reader.readuntil(sep), timeout)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
#########################################]

Verification Complete: SHA = DF9F878419F13509BFA40F1B3D4E919EA5B6DD78
CAN Flash Success

`

closed 2022-12-22 by Johnnyyr

maz0r, 2022-12-22:
> 

The short of this one is that canboot will retry until success (or a number of failures is exceeded).

Typically this error is lack of termination resistors (i.e. your bus isnt 60Ohms) or an otherwise intermittent connection.

With the U2C 2.1 it can also be due to the implementation of candelight on the G0B1.

Johnnyyr, 2022-12-23:
I have changed my setup too Octopus and EBB42 can2can, now it works fine. thx for your description.