maz0r/klipper_canbus #28: Suggestion to flash candlelight on utoc-1/3 results in a non working board

theodiem, 2023-06-08:
Hello,
Thank for the summarization of several pieces together.
I've tried flashing candlelight_fw (using the web updater) and a utoc-1 in DFU mode and it resulted in a non-working board. the status leds don't come up after flash.

Putting the board back into DFU and restoring mellow firmware fixed the board (to the old firmware).

maz0r, 2023-06-08:
Hi @theodiem , 

Can you confirm what mcu model your UTOC-1 has (STM32F072?) and if you used the 2.0 or 0.4 part of the page*? 

The cannable site has been updated since that page was last reviewed and didnt used to have any options (0.4 would be in line with the firmware flashed during the testing I did last year).

Kind regards
maz

theodiem, 2023-06-08:
Hi @maz0r ,

> Can you confirm what mcu model your UTOC-1 has (STM32F072?) and if you used the 2.0 or 0.4 part of the page*?
I have the STM32F072C8T6 (utoc-1) and I've used the cannable 2.0 (https://canable.io/updater/canable2.html) and selecting the "candlelight (ba6b1dd 7/16/21)" from the dropdown.

maz0r, 2023-06-08:
> I can flash the 0.4 if that was what worked for you and try again as I have multiple UTOC-1 boards.

If you can that would be great, that way I can update to point at 0.4 or remove
- 0.4  would be what I used (it was the only option last year), the 2.0 firmware appears to be for CAN FD implementation which isnt used by klipper.

theodiem, 2023-06-08:
I've flashed the "candlelight (68df7d5 1/3/21) BETA" from the 0.4 and it works but no status leds at all.
I've tested a scan for a board (python3 flash_can.py -q) and it can find a board over the canbus.

Would be great if you can get somehow mellow to document the pins for the state (or whatever patch/config they used to build our own firmwares on a more recent commit) but so far I believe it can be closed with the mention to use the 0.4 option and that status/tx/rx led will stop working. Only the 5V is lighted up.

theodiem, 2023-06-08:
>  the 2.0 firmware appears to be for CAN FD implementation which isnt used by klipper.

The thing is .. flashing that firmware don't even show up on lsusb at all. Using dmesg doesn't even mention there was anything plugged on USB. So it is really incompatible with the board, not just lacking klipper support for FD.

maz0r, 2023-06-08:
> The thing is .. flashing that firmware don't even show up on lsusb at all. Using dmesg doesn't even mention there was anything plugged on USB. So it is really incompatible with the board, not just [...]

Yeah did some digging and the 2.0 FD support is compiled for a different family of STM chips (STM32G431) so likely different boot offsets etc.

I'll update the docs calling for 0.4 and mention the LED's 

The pin should be easy to identify, I'll have to dig out a UTOC-1 or 3 from Can mountain though :p

theodiem, 2023-06-08:
Does this makes sense?
[image]
[image]

maz0r, 2023-06-08:
> Does this makes sense? 
yes thanks, it looks like PA1/2, also  dug mine out so i can play with candlelight and poke it when I have the brain.
I'll do the 0.4 update tonight.

commit 460c15fb (2023-06-08): Update u2c.md (#30)

fixed links to use 0.4 canable firmware links closes #28
closed 2023-06-08 by maz0r